francotiveron/Choreo #2: E-Stop Logic

pkelm1, 2020-05-07:
Estop Logic is reversed.  For both the axes and global, if the variable is high, the Estop should be good.  If it goes low, that would indicate and Estopped state

francotiveron, 2020-05-07:
normally emergency signals are inverted in the PLC, but this is not good practice in the UI, where there are not safety implications.
As an example, after a program download and before a start of the PLC, all the signals are normally 0, but the comms with the UI are working, The UI would then assume that the ES are healthy even if they aren't. I would recommend to not wire the physical input for the ES directly to the UI, but rather create additional variables with direct logic (high=ES tripped) and set them in the plc cycle as the input signal inverted.

francotiveron, 2020-05-07:
I decided to invert the signal in the UI app

closed 2020-05-07 by francotiveron