francotiveron/Choreo #13: clear moves on move complete bit

pkelm1, 2020-06-11:
There is a new variable for each axis, "GVL.Axis_xx_Move_Complete", which should clear moves on each axis when Axis_xx_Move_Complete goes high and then also reset Axis_xx_Move_Complete

francotiveron, 2020-06-11:
Unclear
Please provide
1) More details on how the new variable has to be managed (who sets, who resets, when)
2) New PLC program containing the new variables

pkelm1, 2020-06-11:
Whenever any axis completes a move absolute or relative command and the function blocks done bit goes high,  "GVL.Axis_xx_Move_Complete" is set and will remain on until the UI turns it off.  Whenever the UI sees this bit, it should clear the move command from the axis and turn off this bit.

Also just for reference, this bit is also used by the backend for determining "cue complete" but they logic behind how that works is transparent to the front end.  As long as the UI resets it after it goes high, all will work fine.

francotiveron, 2020-06-12:
I think this should be done differently. When the axis completes a move (Relative or Absolute), the PLC should reset the move big (MA/MR_Enable). The  UI only way to reset them should be the Clear button. Please let me know if you see problems with this, as the way you propose is less robust and can more easily lead to status inconsistencies.

francotiveron, 2020-06-12:
Implemented as per previous comment. Please reopen if needed

closed 2020-06-12 by francotiveron

pkelm1, 2020-06-12:
Sorry if you have already done the work, but I agree with your idea and have implemented it on my end.  I didn't realize you were just setting it high and that I could reset it that easily.  I thought the UI might be holding it high for some reason.  This was a very easy change on the back end and it works very well if Twincat turns off the move enable bit once complete