francotiveron/Choreo #15: Pop-up message themeing

pkelm1, 2020-06-11:
It would be great if the pop up warning messages can match the overall theme of the rest of the project.  these are very useful, it is just a little jarring when they look like a different windows application.  Also, it would be great if we can have a more clear indication of a lost watchdog.

francotiveron, 2020-06-12:
Message Popups changed to look more like in the Figma design.
Please have a look and feel free to give more detailed feedback is needed. Look-and-feel UI/UX matters are usually quite subjective

francotiveron, 2020-06-12:
Re-design Popups according to Figma

francotiveron, 2020-06-12:
Please confirm that the wanted theme is the one in Figma

francotiveron, 2020-06-13:
- Info Popup
[image]

francotiveron, 2020-06-13:
Complete

closed 2020-06-13 by francotiveron