francotiveron/Choreo #16: load next cue on cue complete bit

pkelm1, 2020-06-11:
whenever a cue is completed the bit GVL.Cue_Complete will be set to high.  This should clear the current cue that is loaded and load the next cue.  the cue complete bit should then be set low

pkelm1, 2020-06-11:
This is similar in function to the move complete bit but it pertains to an entire cue.  When a cue is complete, the Cue_Complete bit will be set high then as stated above will need to be turned off by the UI when the next cue is loaded.  This can also pertain to presets.  when running a preset if you see the cue complete bit you can deselect the preset and set the cue complete bit low

francotiveron, 2020-06-12:
1) This functionality is already working for the Cue. I re-tested it and it works as per specs

2) I do not think it is a sound idea to use the Cue_Complete for the Preset, it could lead to confusion or  unexpected behaviour (for example, if a preset is run after a cue has completed, there is no way for the UI to determine is the Cue_Complete going high refers to a Cue or to a Preset). I propose to add a Preset_Complete/Preset_Loaded flag pair with similar functionalities but dedicated to the Preset. Please let me know your thoughts

francotiveron, 2020-06-12:
Implemented as per previous comment (Preset_loaded and Preset_Complete to be added to TwinCAT)

closed 2020-06-12 by francotiveron

pkelm1, 2020-06-12:
I will add these two new tags today for presets

pkelm1, 2020-06-12:
These tags are now in my program.  if preset_loaded can go high when a preset is loaded, Twincat will set preset_complete high when it is done.  The UI should then deselect the preset and reset_complete

pkelm1, 2020-06-12:
I just confirmed the cue_complete function and it does indeed work so no worries there.  it would be great if there was an option to autoload the next cue though