francotiveron/Choreo #17: mini axis monitor on show page

pkelm1, 2020-06-11:
I mentioned this in my email, but we need some sort of axis monitor on the show page.  This can probably just go on the bottom of the existing page and be fine.  I emailed you an example of what we are thinking.

francotiveron, 2020-06-13:
I am a little afraid that this format for the monitor could result in something that doesn't fit. Please put some more details on what information you would like to see in the monitor and why (a couple of scenarios of its utility). Maybe I can think to an alternative that is more likeable.

pkelm1, 2020-06-13:
The main reason for this is so the user knows what is happening to the axes at all times and it would be ideal if it was a familiar format to the main page that they are used to.  This is not anything that needs the jogging or move functionality of the main screen but really just position and status.  we designed it like what I proposed so it looked familiar, but if you have a better idea just let me know.  currently, I think we still have enough room for a scrolling list of cues positioned above this on the 10" screen

francotiveron, 2020-06-14:
Implemented as per request

[image]

closed 2020-06-14 by francotiveron