francotiveron/Choreo #19: preset in position illumination

pkelm1, 2020-06-12:
this is something we haven't talked about before but it would be great if there was a way for UI to check the positions saved in a preset against the actual positions of the axes and somehow illuminate the preset button if it was in position.  basically a way of letting the operator know which preset the room is currently in.

francotiveron, 2020-06-13:
Should this work only on the selected preset or for all (for example, if the positions match those of a dark-blue preset, should it illuminate)?

pkelm1, 2020-06-13:
In an ideal world, this would happen for all presets, even those that are not selected.  The idea is that a user might look at the controller not knowing what the previous user did and quickly know which preset position the room is in of any

francotiveron, 2020-06-13:
Due to usual imprecision in real world. It is impossible that the axes will be in the precise position, so this feature needs to define an "acceptability margin" within which an axis can be considered in-position. What is an acceptability window? (e.g. +/- 1 inch)? Of course if the window is too strict it may not illuminate the preset, if it is too wide it could be illuminated also when it shouldn't. You know better the machine, please tell me what is an optimal value.

francotiveron, 2020-06-13:
Implemented with border coloration and margin 0.1 inch
[image]

closed 2020-06-13 by francotiveron