francotiveron/Choreo #20: Axis selection alignment in moves and cues

pkelm1, 2020-06-12:
I know you said the next version has the fix for not being able to select grouped axes, so that will be great.  we would also like to see if there is a way to keep everything aligned a little better.  currently if we rename an axis its alignment to the rest of the axes in the selection page is off depending on the numbers of characters

francotiveron, 2020-06-13:
Couldn't see any misalignment in my tests. Please add screenshots

pkelm1, 2020-06-13:
I am not at my computer right now, but on mine if I change a couple of axis names, then go into a move command, the axis names on the bottom where you can select other axes are out of alignment since they are have different length names

francotiveron, 2020-06-13:
Fixed
Changed check box alignment in  motion page from center to left

closed 2020-06-13 by francotiveron