francotiveron/Choreo #23: can write cues for axes in groups

pkelm1, 2020-06-12:
this may be fixed in your next implementation already, but just like the move window, I can write a grouped axis into a cue and shouldn't be able to.  if this is already fixed, no worries

francotiveron, 2020-06-13:
Cues are not like moves, they are "forever". As an example. you could create a cue containing axis 1 when it is not grouped and then, later, put the same axis 1 in a group, yet the cue is already there and doesn't make sense to remove it from the database. It would be better to leave the cue editing independent on the current axis grouping. The real point when the problem occurs is when we are going to execute (download them).  One way to fix  is to extend the Status (for example now we have Enabled and Disabled, we could add Invalid). An Invalid cue would be skipped by the loader like a disabled one.

pkelm1, 2020-06-13:
I think the invalid state is a good idea.  My main request here was to make cue writing like moves with respect to what is visible.  If you try to write a cue and axes 1 and 2 are assigned to a group, you shouldn't be able to select them since you have to move the group as a whole

francotiveron, 2020-06-14:
New feature "Cue Inconsistency" implememented
[image]

- In the Show page, Cues and Cue Rows containing non actionable aces (motors that are grouped, groups or motors that are not user-enabled), will appear with a red border (inconsistent)
- Inconsistent cues can still be loaded manually, but the automatic load (on cue complete) will skip them even if they are Enabled.

closed 2020-06-14 by francotiveron