francotiveron/Choreo #26: Set new position can only select text

pkelm1, 2021-08-13:
in axis setup, you have to actually click on the text of set new position as opposed to pressing anywhere on the button.

francotiveron, 2021-08-14:
Default Button Background was null, changed to Black

pkelm1, 2021-09-13:
This isnt a bug for this fix but was discovered when testing it.  if a groups position is changed in the setup screen and then the user presses cancel to exit the setup screen the new position is kept but the groups offset to the other axes is not saved.  I think a good fix for this would be to have a confirmation pop up confirming the user wants to set a new position and upon pressing yes it should toggle the set position variable like it did before but also toggle the new group save variable just like pressing save on the settings page.

it might also be useful to have the confirmation pop up on the axis pages as well but the additional functionality for the group save is not required for an axis

closed 2021-10-02 by francotiveron