francotiveron/Choreo #27: can still assign motors in a group to a cue (maybe gray them out instead of remove)

pkelm1, 2021-08-13:
When writing a cue, it is possible to select a grouped motor rather than needing to select the whole group.  this has been resolved in other places but just seems to have been missed in cue creation

francotiveron, 2021-08-14:
This is by design:
When editing/creating a cue, there is no limitations, because the cue can be used in configurations different from the current.
However, in the Show page, cues that are inconsistent with the current configuration of the system are displayed with red border, so the user knows that the current system configuration is not compatible with that cue.

pkelm1, 2021-08-31:
I am not really against it just being disabled, but in my current testing, I have found that I have disabled cues and no real easy way to know why or get them functional again.  Normally, I delete then start over but this may not always be desirable. 

Is there a way to see what is bad in a cue?

francotiveron, 2021-09-04:
[image]

closed 2021-10-02 by francotiveron