francotiveron/Choreo #28: change look of jog velocity slider and label

pkelm1, 2021-08-13:
The jog slider looks out of place with the rest of the graphics.  Not sure how hard it would be make it just look like the position bar on the axes rather than looking like a basic windows form

francotiveron, 2021-08-14:
[image]

Please confirm that you mean as per the pcture

pkelm1, 2021-08-14:
We don't like the look of the jog slider.  If it can just be a solid bar like the position bar you have highlighted I think it would look better.   It should keep the same functionality the slider has, just change the lookPaul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.*** 

  label (#28)

francotiveron, 2021-08-15:
Changed to ScrollBar
[image]

closed 2021-10-02 by francotiveron