francotiveron/Choreo #30: Toggle GVL.Group_XX_save when saving a group

pkelm1, 2021-08-13:
when pressing either save button during group edits, I added new variables GVL.Group_XX_save that should be toggled for whatever group is being saved where XX is replaced by the group number, 01, 02, etc

francotiveron, 2021-08-14:
Please send the updated TwinCAT project

pkelm1, 2021-08-14:
The current project is kind of a shit show of changes.  I am in the process of cleaning everything up, give me a day and I will get it overPaul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.*** 

  group (#30)

francotiveron, 2021-08-21:
@pkelm1 
Remember that, as other pulse command, resetting the flag is a PLC program concern

pkelm1, 2021-08-21:
Yep, I can resetPaul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.*** 

  group (#30)

closed 2021-10-02 by francotiveron