francotiveron/Choreo #31: Motor rotations per ft should also write to GVL.Axis_XX_Rotations_Per_Foot

pkelm1, 2021-08-13:
I added new variables GVL.Axis_XX_Rotations_Per_Foot for each axis and these should be written to when the value is saved on the axis setup screen.  Also, unless you need it for something, we can get rid of the motor rotations per foot for groups since they are always sim axes and dont actually have a motor.  if need be, we can just make that value 1, but please let me know what you end up doing.

francotiveron, 2021-08-21:
Waiting issue from QuickConverter https://github.com/JohannesMoersch/QuickConverter/issues/14

francotiveron, 2021-08-21:
Waiting issue from QuickConverter https://github.com/JohannesMoersch/QuickConverter/issues/14
In alternative, duplicate focusable grid and hide binding to IsGroup/IsMotor

francotiveron, 2021-09-04:
Help didn't come => controls duplicated

closed 2021-10-02 by francotiveron