francotiveron/Choreo #33: Limit a groups max velocity to the highest velocity of any slave axis

pkelm1, 2021-08-13:
Not sure how difficult this is, but it would be great if a groups max velo, accel and decel couldnt exceed any of the axes in that group

francotiveron, 2021-08-14:
Please clarify:
1) Should group max be limited to axes maxes or to axes defaults?
2) What about min?
3) What should happen to the group values when motors are removed and determine a change in the limits?

pkelm1, 2021-08-14:
Group max should be determined by axis max.  Min should also apply.In an ideal world if a slower motor is added to a group, the groups max would be dropped to match.  If the slowest motor is removed, then it can be up to the user to readjust Paul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.*** 

  velocity of any slave axis (#33) 
Please clarify:

What about min?

francotiveron, 2021-08-21:
@pkelm1 

I can think of the following uncovered scenarios

1) User add motor whose max is lower than the min of another motor (in the same group)
2) User changes max or min of the motor that already defines the group max or min
3) User adds a motor that determines a change in the max or min of the group. At this point, the group max/min has changed in the UI but not in the PLC. The user needs to remember to go into the group page and save
4) A group config is saved in one of the 10 stored configuration (as per other requests). Then the user changes the max or min of one of the motors that belong to that group and this changes the max/min for the group. Finally the user loads the config he saved previously, but the max/min is different than the actual

There could be other critical scenarios. Please:
- Think to other scenarios that I could have missed
- Specify the behaviour for all scenarios

pkelm1, 2021-08-31:
Can we just not allow the user to change any of the values of a grouped motor.  in which case, they would have to remove it from the group, change it, then re add it to the group.  This would then require a group save which would cover most scenarios above.

As for changing saved group configs, since my idea of loading configs only populated the data on the configuration page and it is saved when the top save button is pressed, would it be possible to just lower the max or raise the min if the saved config is our of range.  For example, saved config has a max of 16 but the fastest motor is 12.  When loading the config, all values populate but the max is either listed in red or automatically changes to 12

closed 2021-10-02 by francotiveron