francotiveron/Choreo #34: ability to save axis configs

pkelm1, 2021-08-13:
This isn't a punch note per se but a new feature.  it would be great if there was a way to save an axis and be able to select them from a drop down.

This might be part of a larger conversation about saving show files and/or an entire controller config file.

francotiveron, 2021-08-14:
Please provide more context
1) What axis data should be saved
2) Where should the drop-down be located in the interface?

Note that config is already saved in the windows registry (loaded at application startup). 
I reckon the request is for having multiple configurations saved with a name and then reloadable. If so, other consideration come to light (ex. backup, transfer, copy...)

pkelm1, 2021-08-14:
The idea would be that it saves everything on the axis setup screen excluding the set new postion value.  Basically this system will typically be used with a few different axes that are all the same configuration.  I could see a user setting up the first one and saving it then just selecting the same configuration to setup then next few axes.  If we could have 10 or so configs that should cover most casesPaul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.***

francotiveron, 2021-08-21:
@pkelm1
Please tell me if this rough design would be acceptable
[image]
1) Click on one of the 10 available buttons on the left (10 stored configuration)
2) This shows the values of the selected configuration on the right panel
3) SAVE button on the right panel will take the current config and save it on the selected configuration
4) LOAD would  tale the saved configuraiton and override the current values

pkelm1, 2021-08-21:
It would be better if rather than number we could have them named.CM chain motorChain MasterPF winchEtc.Is it possible to do a drop down rather than select buttonsPaul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.***

pkelm1, 2021-08-31:
[image]

I don't think it has to be very complicated, just a drop down and a load and save button.  Basic order of operations would be:

 - Configure your axis as normal then press the new "Save" button for the configs
 - Keyboard pops up to give the configuration a name
 - Save the new name and the new configuration will be available in the drop down

Otherwise you can select a different configuration and press load.  Pressing load only has to populate the axis values on the setup screen and they can just be saved to twincat when they always have when the user presses the "save" button on the top of the screen to exit the setup page

closed 2021-10-02 by francotiveron