francotiveron/Choreo #48: add an analog jog stick

francotiveron, 2022-02-04:
We would like to add an analog jog stick to this system which would require a slight change to the swiping action for an axis.  If we change it so swipe up/down stays the same, swipe right goes into the move window as it currently does and swipe left sets a jog stick enable variable high and the axis displays “Jog Stick”, I can handle everything else in the back end.  In order to do a relative more, we will just swipe right and select relative from the move page which we can already do.

francotiveron, 2022-02-04:
I assume than the "Joy Stick" display would replace the "Ok" at the bottom of the axis card in the main page. Please confirm

Please send me a copy of the PLC program containing this feature

pkelm1, 2022-02-05:
It should display "jog Stick"

francotiveron, 2022-02-19:
I'd like to be sure that this has been carefully considered. Please think about these doubts I have:
1) How would the Joystick binding be reset? 
2) What would happen to the group, preset, cueing features when one axis is under joystick?
3) Would groups also be subject to joystick?

pkelm1, 2022-02-19:
1. This would get reset just like any other.  Swipe up to jog up, swipe down to jog down, swipe left to jog stick enable, swipe right to move enable, swipe the opposite way or press clear to clear.  There is still a physical button to enable the stick so it doest just start running once enabled.

2. This would be behave just like every other move.  You can jog stick enable multiple axes but you can't stick enable a grouped axes.  You can stick enable a group of axes though.  Only caveat would be that it be nice in general if you can't stick or traditional jog enable 2 axes with different max/min velocity 

3. Yep

francotiveron, 2022-02-26:
Please specify in more detail the sentence 
**"Only caveat would be that it be nice in general if you can't stick or traditional jog enable 2 axes with different max/min velocity"**
How should the application exactly behave?

francotiveron, 2022-02-28:
from Paul:
The current problem with jogging is it currently does a percentage of full speed which works fine for multiple axes when they have different max speeds.  The problem is that we are doing jobs with different types of machinery so a user might expect 2 machines to jog together but they have different maxes so they move different speeds.

There are already pop ups that warn if we try to break another rule like trying to move a grouped axes by itself.   Can we just do a pop-up that says something like "only axes with equal velocity profiles may jog together"

So you know the plan on my end is for the swipe up/down jogging to work the same with the speed slider.  If the jog stick is assigned, the axis speed will depend on the stick with the max being determined by the slider.  I will handle all that in twincat.

closed 2022-05-14 by francotiveron