francotiveron/Choreo #84: following error enable check box and value

francotiveron, 2022-07-30:
I cant remember if I asked for a following error enable check box and value but it would be nice to have as well as the soft limits.

francotiveron, 2022-07-30:
@pkelm1 
If my understanding is correct, you want another check box and two additional values in the settings page, like
[image]

If so please

- Confirm Page layout
- Specify measurement unit and limits for he new value
- Provide TwinCAT program with required values

francotiveron, 2022-09-03:
From Paul email
`
I just looked and the following error variables were already in the twincat program:

Axis_01_Follow_Error_Enable: BOOL;

Axis_01_Following_Error: LREAL;
`

Please answer the two remaining questions from previous comment

- Confirm Page layout
- Specify measurement unit and limits for he new value

pkelm1, 2022-09-03:
There can be another check box the enable following error by soft limit and load cell enable.  As for the location of the data field, I am not near a system right now to look but if you can propose somewhere I can take a look.  Units should be feet and inches Paul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.*** 

  (Issue #84)

francotiveron, 2022-09-03:
[image: image.png]

francotiveron, 2022-09-10:
From Paul email

'layout is fine'