francotiveron/Choreo #85: cues values beyond the max accel, vel or decel

francotiveron, 2022-07-30:
From Paul Email"

'
Not a new note but one I just noticed.  It is possible to enter values in cues that are beyond the max accel, vel or decel value
'

francotiveron, 2022-07-30:
@pkelm1 
This is by design:
- Cues can contain many axes, with different limits. Hence there is not a max value for a cue Even if we were to take the min max for the current selected axes, the user could later add an axis with a lower max which would pose a similar problem.
- Axes limits can be changed independently of cue parameters. Even if we limit the values in the cue, nothing prevents the user to change a cue axis limit, making the cue non-max-compliant.

The software already checks cues in the Show page. For example, if a cue encompasses a disable axis, the cue is shown with a red border and an explanation is given. This check doesn't currently evaluate limits, but it could be done. However, this check is in the Show page not when entering cure parameters.

You said you want to review the cueing system, maybe this inconsistency can be taken into account in designing the new way. 

In any case, please come back to me with your comments.

francotiveron, 2022-09-03:
Please confirm the previous comment makes sense to you and I can close this issue

pkelm1, 2022-09-03:
Let not worry about this one for this releasePaul KelmOwner / CEOMain: 410-231-7214, Direct: ***@***.*** 

  decel  (Issue #85) 
Assigned #85 to @pkelm1.

closed 2022-09-03 by francotiveron